qgallouedec/panda-gym #1: How to load the franka panda with colours in pybullet?

estellexky, 2021-03-13:
Hi! I would like to ask how to load the franka panda in pybullet with white and black colours (just like you showed in readme). I have tried the franka_panda_description package in your github, however I got the franka in pybullet with only white colour.

Looking forward to your reply. Thanks a lot.

qgallouedec, 2021-03-13:
Hi Estelle, thank you for your question!

If you installed the package as explained in the `README.md`, the franka panda should load without having to install any additional packages. Do the following command lines give you an error?

```bash
pip install panda-gym
```
Then, in a python script:
```python
import gym
import panda_gym
env = gym.make('PandaReach-v0', render=True)
```

----

If your question is more general and does not concern the use of my package, here is how to load the panda (without having to install any additional package) :

```bash
pip install pybullet
```
then, in a python3 script:

```python
import pybullet as p
import pybullet_data
p.connect(p.GUI)
p.connsetAdditionalSearchPath(pybullet_data.getDataPath())
p.loadURDF("franka_panda/panda.urdf")
```
Result:

[image]

estellexky, 2021-03-13:
I can load successfully panda now with your instruction. It seems that I didn't know franka_panda/panda.urdf already exists. 

Thanks a lot!

closed 2021-03-13 by estellexky