qgallouedec/panda-gym #3: Problem

ZY-KK, 2021-06-22:
Hello, first of all, thank you for the work, but I have a question. When I create the robot and make robot grasp a object, the gripper will pass through the object, and the contact point detect is zero, could you please tell me why I met this?

qgallouedec, 2021-06-22:
Hi,
You should not encounter this behavior. Can you send a screenshot of the rendering? Can you publish the code to reproduce this behavior?

ZY-KK, 2021-06-22:
I add a object in certain position, and when I try to control robot to grasp it , it pass through the object, and also the robot even can pass through the table I add, did I add wrongly? I just add object by using p.loadURDF

ZY-KK, 2021-06-22:
[image]
as the picture shows, the finger actually pass through the object and if I try to lift it, object will not be lifted

ZY-KK, 2021-06-22:
> Hi,
> You should not encounter this behavior. Can you send a screenshot of the rendering? Can you publish the code to reproduce this behavior?

qgallouedec, 2021-06-22:
Hi,

In order for me to work on this issue, I need to be able to reproduce it. The first step is to publish the **minimal** code that allows to reproduce it. (This is a good practice in solving issues on Github.)

With the little information you gave me, I wrote the following code, for which I don't observe any collision problem.

```python
import panda_gym
from panda_gym.pybullet import PyBullet
from panda_gym.envs.robots.panda import Panda

sim  = PyBullet(render=True)
panda = Panda(sim, base_position=[0.0, 0.0, 0.6])
panda.reset()
sim.loadURDF(body_name='table', fileName='table/table.urdf', basePosition=[0.0, 0.0, 0.0])

while True:
    sim.step()
    panda.set_action([0.00, 0.00, -0.02, 0.00]) # going down, by 2 cm per step
    sim.render()
```

[image]

Could you send me the minimal code for which you observe this problem?

ZY-KK, 2021-06-22:
Thanks for your answer, I do not know the reason, but now it works, maybe because I modify some code. One question, when run the code, actually robot move slowly, could you please tell me how can accelerate the move speed?

closed 2021-06-22 by qgallouedec