qgallouedec/panda-gym #5: The newest version 1.1.0 provide the new random seed, what does it mean?

chenxing618, 2021-08-19:
I found that the author pubulished the newest version 1.1.0. In the release introduction, it shows that the newest version accomplish correct implementation of a random seed for reproducibility. I do not get what the author means? And Comparing to the 1.0.1, which exact problem can the 1.1.0 fix based on this new advantage?

qgallouedec, 2021-08-19:
Hi,
For versions prior to 1.1.0, it was not possible to manually set the random seed. Here is an example: 

```python
import gym
import panda_gym # version 1.0.1
env=gym.make("PandaReach-v1")
env.seed(123)
obs1 = env.reset()
env.seed(123)
obs2 = env.reset()
# here, obs1 != obs2
```

From version 1.1.0 onwards, it is now possible to do this, and thus to strictly reproduce results.

```python
import gym
import panda_gym #  version 1.1.0
env=gym.make("PandaReach-v1")
env.seed(123)
obs1 = env.reset()
env.seed(123)
obs2 = env.reset()
# here, obs1 == obs2
```

I hope I answered your question.

closed 2021-08-19 by qgallouedec

chenxing618, 2021-08-19:
Many thanks for that! I am clear for this question