qgallouedec/panda-gym #6: Set joint angle not change position

yzhao334, 2021-08-20:
When resetting environment, robot joint angle seems not back to predefined neutral position. Is that because by default velocity control mode is used? Instead of using resetJointState function, how about using setJointMotorControlArray function, setting control mode to be POSITION_CONTROL and set the desired position?

qgallouedec, 2021-08-24:
Hi @yzhao334 , thank you for reporting this bug.

Can you provide me a minimal code to reproduce this bug? (it's a good practice when you open issues on GitHub)

I tried to reset the environment and the robot comes back to its initial position every time: 

```python
import gym
import panda_gym

env=gym.make('PandaReach-v1')

# initial reset
obs = env.reset()
print(obs['observation']) 
#[ 3.84396701e-02 -2.19447219e-12  1.97400143e-01  0.00000000e+00 -0.00000000e+00  0.00000000e+00]

# make a litlle move 
obs, _, _, _ = env.step([0.5, 0.5, 0.5])
print(obs['observation']) 
#[0.05003328 0.0178855  0.21331219 0.25930479 0.34940061 0.3846675 ]

# reset
obs = env.reset()
print(obs['observation'])
# [ 3.84396701e-02 -2.19447219e-12  1.97400143e-01  0.00000000e+00 -0.00000000e+00  0.00000000e+00]
```

yzhao334, 2021-08-24:
Hi @qgallouedec, thanks a lot for replying. I was enabling real time simulation of pybullet for realistic object movement, such as dropping the cube when not grasping properly. I have found that the default setting does not show this issue, but the robot does not return to the default initial position when real time simulation is enabled. 

I am attaching the code as below.

```python
import gym
import panda_gym
import pybullet as p

env=gym.make('PandaPickAndPlace-v1',render=True)

# enable real time simulation for more realistic object interaction
p.setRealTimeSimulation(True)

# initial reset
obs = env.reset()
print(obs['observation']) 

# make an obvious move 
for _ in range(30):
    obs, _, _, _ = env.step([0.5, -0.5*5.0/3.0, 0, 0])
print(obs['observation']) 

# reset
obs = env.reset()
print(obs['observation'])
```

qgallouedec, 2021-08-24:
To make the rendering more realistic, **you should not use** `p.setRealTimeSimulation(True)` but `env.render()` after each `env.step(action)`:

```python
for _ in range(30):
    obs, _, _, _ = env.step([0.5, -0.5*5.0/3.0, 0, 0])
    env.render() # waits just the right amount of time to make the render realistic
print(obs['observation']) 
```

Explanation :
By default, pybullet waits for `p.stepSimulation()` to step the simulation. I use it in this package. If you run `p.setRealTimeSimulation(True)`, pybullet won't wait for `p.stepSimulation()` to step the simulation. It will step _whenever it wants_ (see [Quickstart guide](https://usermanual.wiki/Document/pybullet20quickstart20guide.479068914/html)). And this is not what we want, since it makes the behaviour random (depending on the power of your computer among other things).
Instead, to have a realistic rendering, we simply make a pause that corresponds to the simulation step. This pause is made when you call `env.render()`. (See the [source code](https://github.com/qgallouedec/panda-gym/blob/d32eb4b59d41a28acef34d9be62f4deeff7f5c6f/panda_gym/pybullet.py#L90))

closed 2021-08-24 by qgallouedec