qgallouedec/panda-gym #9: Why did the run window close soon，just few seconds

learningxiaobai, 2021-09-08:


qgallouedec, 2021-09-08:
Hello @learningxiaobai, 

I don't understand your problem. Can you be a bit more specific and provide at least a description in addition to the title? 

Here are some good practices when you will open your future issues:

- Explain the expected behaviour
- Explain the actual behaviour,
- Provide a minimal code to reproduce the bug, and the logs if any,
- Provide your system specification (package version, python version, OS)

I will be happy to help you once I have this :)

learningxiaobai, 2021-09-09:
> Hello @learningxiaobai,
>
> I don't understand your problem. Can you be a bit more specific and provide at least a description in addition to the title?
>
> Here are some good practices when you will open [...]

hello，your codes are good and i ran well(no bugs),
 1，2：my expected behaviour is that the robot can finish the task(such as push or grasp) through many trainings and the run window will stay until you close it,but the actual behaviour is that the run windows just stay few seconds and it shut down soon,
3.the logs are：
MotionThreadFunc thread started
numActiveThreads = 0
stopping threads
Thread with taskId 0 with handle 00000000000005EC exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 0000000000000364 exiting
Thread TERMINATED
4.package version is 1.1.0，my python version is 3.6.8，windows
Sorry, I replied to you late because of the jet lag，
hope you can solve my problems,thanks,

learningxiaobai, 2021-09-09:
> Hello @learningxiaobai,
>
> I don't understand your problem. Can you be a bit more specific and provide at least a description in addition to the title?
>
> Here are some good practices when you will open [...]

learningxiaobai, 2021-09-09:
[image]
just like this

closed 2021-09-09 by learningxiaobai

qgallouedec, 2021-09-09:
The behavior is as expected. 

1. A window opens because rendering is enabled (`render=True`).
2. An action is randomly chosen :`action = env.action_space.sample()`.
3. Then, after 50 timesteps, the environment returns `done=True`, so the loop stops. 
4. The environment is closed :`env.close()`, and so is the rendering window. 

This directory does not propose any handwritten algorithm for the realization of the different tasks proposed. The whole point is that they are learned by reinforcement. See [rl-baselines3-zoo](https://github.com/DLR-RM/rl-baselines3-zoo) directory, which proposes a well supplied benchmark of algorithm with the `panda-gym` tasks. You will be able to load pre-trained policies that you will find in the directory [`rl-trained-agents`](https://github.com/DLR-RM/rl-trained-agents/).

learningxiaobai, 2021-09-10:
> The behavior is as expected.
>
> 1. A window opens because rendering is enabled (`render=True`).
> 2. An action is randomly chosen :`action = env.action_space.sample()`.
> 3. Then, after 50 timesteps, the [...]

thanks a lot,best wishes to you