qgallouedec/panda-gym #17: AttributeError: module 'pybullet' has no attribute 'GUI'

JakobThumm, 2022-01-04:
## Describe the bug

I cannot make a gym environment anymore.
I get an error stating
```
Traceback (most recent call last):
  File "test.py", line 26, in <module>
    env = gym.make('PandaReach-v2', render=True)
  File "/home/jakob/anaconda3/envs/panda/lib/python3.7/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/home/jakob/anaconda3/envs/panda/lib/python3.7/site-packages/gym/envs/registration.py", line 129, in make
    env = spec.make(**kwargs)
  File "/home/jakob/anaconda3/envs/panda/lib/python3.7/site-packages/gym/envs/registration.py", line 90, in make
    env = cls(**_kwargs)
  File "/home/jakob/Promotion/code/panda-gym/panda_gym/envs/panda_tasks/panda_reach.py", line 20, in __init__
    sim = PyBullet(render=render)
  File "/home/jakob/Promotion/code/panda-gym/panda_gym/pybullet.py", line 34, in __init__
    self.connection_mode = p.GUI if render else p.DIRECT
AttributeError: module 'pybullet' has no attribute 'GUI'
```
## To Reproduce

Starting on a clear plate
```
conda create -n panda python=3.7
conda activate panda
~/panda-gym$ pip install -e .
```
and running the quick start code 
```
import gym
import panda_gym

env = gym.make('PandaReach-v2', render=True)
```
leads to the aforementioned error. 
I also tried python 3.8

## System

  - OS: Ubuntu 20.04
  - Python version (`python --version`): 3.7, 3.8
  - Package version (`pip list | grep panda-gym`): panda-gym 2.0.0 | pybullet 3.2.1

qgallouedec, 2022-01-04:
Hello,

Thank you for reporting this bug. 

Can you try to execute the same python code from `~` ?

```bash
cd  ~
python <PATH_TO_YOUR_CODE>/your_code.py
```

Intuition:
After checking, the latest version of pybullet still has the GUI attribute. So I think this bug comes from the fact that you are running this script from a `panda-gym/panda_gym` At this location `pybullet` is referring to `pybullet.py` (a panda-gym helper file) and not the `pybullet` module.

JakobThumm, 2022-01-04:
Ah, this worked. 
Thank you!

closed 2022-01-04 by JakobThumm