qgallouedec/panda-gym #18: I trained panda push using sb3, however i got this error when loading the model. Observation spaces do not match

isaacncz, 2022-01-14:
raise ValueError(f"Observation spaces do not match: {observation_space} != {env.observation_space}")
ValueError: Observation spaces do not match: Dict(achieved_goal:Box([-10. -10. -10.], [10. 10. 10.], (3,), float32), desired_goal:Box([-10. -10. -10.], [10. 10. 10.], (3,), float32), observation:Box([-10. -10. -10. -10. -10. -10.], [10. 10. 10. 10. 10. 10.], (6,), float32)) != Dict(achieved_goal:Box([-10. -10. -10.], [10. 10. 10.], (3,), float32), desired_goal:Box([-10. -10. -10.], [10. 10. 10.], (3,), float32), observation:Box([-10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10.
 -10. -10. -10. -10.], [10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10.], (18,), float32))

MH-Hou, 2023-08-31:
Hi,

I kind of encountered a very similar problem when I tried to load a trained model from HuggingFace for PickandPlace-v1. Could you please tell me how you solved this problem of observation space mismatch?

qgallouedec, 2023-08-31:
Can you share the code used ? What version of sb3 and panda-gym do you use?

MH-Hou, 2023-09-01:
Thanks for your quick reply!

Below is the code I ran:
`test_env = gym.make("PandaPickAndPlace-v1", render=False)`
`checkpoint = load_from_hub(repo_id="sb3/tqc-PandaPickAndPlace-v1", filename="tqc-PandaPickAndPlace-v1.zip")`
`model = TQC.load(checkpoint, env=test_env)`

After I ran the code in the terminal, it complained that
`raise ValueError(f"Observation spaces do not match: {observation_space} != {env.observation_space}")
ValueError: Observation spaces do not match: Dict('achieved_goal': Box(-10.0, 10.0, (3,), float32), 'desired_goal': Box(-10.0, 10.0, (3,), float32), 'observation': Box([-10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10. -10.
 -10. -10. -10. -10. -10.   0.], [10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10. 10.
 10.  1.], (20,), float32)) != Dict('achieved_goal': Box(-10.0, 10.0, (3,), float32), 'desired_goal': Box(-10.0, 10.0, (3,), float32), 'observation': Box(-10.0, 10.0, (19,), float32))
`

For your information, I used the stablebaselines3 of version 2.1.0 and panda_gym of version 1.1.1

I also tried to load the model you uploaded to the huggingface by using the code:
`test_env = gym.make("PandaPickAndPlace-v1", render=False)`
`checkpoint = load_from_hub(repo_id="qgallouedec/tqc-PandaPickAndPlace-v1-4094880237", filename="
tqc-PandaPickAndPlace-v1.zip")`
`model = TQC.load(checkpoint, env=test_env)`

Then this time it complained that 
` raise KeyError("The observation_space and action_space were not given, can't verify new environments")
KeyError: "The observation_space and action_space were not given, can't verify new environments"
`

Could you please tell me what I should do to solve these issues?

closed 2023-10-24 by qgallouedec