qgallouedec/panda-gym #19: Can not reproduce the environment by the latest gym

Kami-code, 2022-01-28:
I cannot simply reproduce what you described in installation tutorial. Each time I run, I will get the error in the image.

It seems the latest Gym has removed something about GoalEnv which leads to the error.  It seems to be related with the [latest update](https://github.com/openai/gym/pull/2516/commits/dea1916095c0a587c24b0d070669fed31f652a8a). Can you fix it? Or add some requirement of the gym version.
[image]

qgallouedec, 2022-01-28:
Hi, thank you for reporting.
Indeed, `GoalEnv` has been removed from dev version of Gym. But the current stable version installed with `pip` is 0.21.0. If you install gym with `pip` you shouldn't encountered the error. Try

    pip install gym==0.21.0

I leave this issue open to remember that the next gym version will require an update of the dependencies.

Kami-code, 2022-01-28:
It works. Thanks a lot.

closed 2022-02-28 by qgallouedec