qgallouedec/panda-gym #22: Feature idea: multiple Panda robot in the same environment

meinczingerm, 2022-04-02:
I am currently implementing an environment similar to the cable insertion task presented in https://arxiv.org/pdf/2112.00597.pdf, but I have an issue that I can not add and control more than 1 Panda robot at the same time. I managed to resolve this issue in a hacky way, but since it is fairly simple to solve and could be useful for others I wanted to create this issue.

The problem is caused by the same body_name="panda" of all added robot:
This could simply be fixed by passing an ID or something which could be added at the end of the body_name.

It would be also nice to pass a robot orientation similar to base_position, so the robots can be placed with different orientation.

qgallouedec, 2022-04-04:
That's sounds like an interesting enhancement. 

Can you open a draft PR with your fixes, and post a code example to create an environment with two robots ?
How do you handle the action space ? And the observation space ?

The task part should not be a problem because the task and the robot are independent.

meinczingerm, 2022-04-04:
I created a PR: https://github.com/qgallouedec/panda-gym/pull/23

This is my first contribution so feel free to give me any feedbacks.

closed 2023-10-24 by qgallouedec