qgallouedec/panda-gym #26: SB3-zoo has models for v1 tasks but only registration specs for v2 tasks

dirkmcpherson, 2022-05-09:
Hi, I'm trying to get a SB3 model to train the harder tasks (so far I've failed with SAC+HER), so I went to SB3-zoo to see some examples of successful models. I can't get them to load, and it looks like its because zoo has trained models for the v1 versions, but only has the v2 environments registered. Do you have successful trained V2 models you can push to zoo / did you successfully train any SB3 models out of the box on the v2 versions of the task? 

thanks for making this environment!

dirkmcpherson, 2022-05-09:
FYI you can get around this by replacing the "1" with a "2" in the env_id after the model is loaded in enjoy.py for the "ExperimentManager.is_atari" check and then in the "create_test_env" call.

qgallouedec, 2022-05-10:
Indeed.

First of all, this issue would be more appropriate in sb3-zoo than here. I will answer it anyway.

I am updating the sb3-zoo baselines for the v2 version of panda-gym. I should open a PR on sb3-zoo soon. I will put the link to the PR here when it is opened.

For the tasks already present in v1, the changes are quite small : it's mostly about some physical parameters like friction. So it should be possible to just replace v1 by v2 in the model loading without major trouble.

Be careful though, if you do that, you will necessarily get worse results than if you use a model trained with v2

Translated with www.DeepL.com/Translator (free version)

qgallouedec, 2022-05-13:
https://github.com/DLR-RM/rl-baselines3-zoo/pull/244

closed 2022-05-13 by qgallouedec