qgallouedec/panda-gym #30: Real Life Implementation on Franka Emika Panda

ruturajsambhusvt, 2022-05-31:
I am working to implement reach task (and trajectory tracking then) on a real Panda robot.  I am afraid that I am quite new to this field and am the first in my lab working on learning (my lab focuses on design and control theory). I was following this paper - https://arxiv.org/abs/1803.07067

It is recommended here to use Velocity control for low level joint control.
I was also reading up your post - https://gallouedec.com/post/panda-gymv0/ which outlines the limitations on sim to real transfer.

It would be great if you would be able to guide me on this path.

qgallouedec, 2022-05-31:
This sounds interesting,
Honestly, I have no experience with sim2real transfer, I just know that it is not straightforward. 

First point, I understand that you want to work with velocity control. panda-gym works with velocity control: actions are velocity targets. So you don't need to modify the code of panda-gym.

Secondly, you should configure the interface with gym and ROS to use the learned model on the real robot. I wrote a code in the past that may help you: https://github.com/qgallouedec/franka_python . It is an interface between the real panda robot and python via ROS.
It is provided as is, I no longer maintain this repository.

If you find/write a clean and nice code that allows to control a real robot based on panda-gym, let us know!

ruturajsambhusvt, 2022-05-31:
Thank you for the response. I was looking into PyBullet documentation, and I guess there is this line of code here in the PyBullet.py ->
 self.physics_client.setJointMotorControlArray(
            self._bodies_idx[body],
            jointIndices=joints,
            controlMode=self.physics_client.POSITION_CONTROL,
            targetPositions=target_angles,
            forces=forces,
        )

I think this needs to be changed to VELOCITY_CONTROL, please correct me if I am wrong. I did that and am trying to train the reach joint space sparse reach task, but there is no much improvement after 5.5 million timesteps.

qgallouedec, 2022-05-31:
Actually, it is indirect velocity control since the target position is `current_position + action*dx`. 

See this line:

https://github.com/qgallouedec/panda-gym/blob/68c87420fa1ced96a52bae3eef2dd596fd4d820c/panda_gym/envs/robots/panda.py#L84

I don’t know if it is equivalent to velocity control in pybullet.

closed 2022-06-28 by qgallouedec