qgallouedec/panda-gym #38: train_push seems to be broken after merging gymnasium branch

mfouadthabet, 2022-10-23:
## Describe the bug

The following error occurs on trying to run train_push

assert not hasattr(obs_space, "spaces"), f"Unsupported structured space '{type(obs_space)}'"
AssertionError: Unsupported structured space '<class 'gymnasium.spaces.dict.Dict'>'

## To Reproduce

Provide a **minimal** code :

```python
import gymnasium as gym
from stable_baselines3 import DDPG, HerReplayBuffer

import panda_gym

env = gym.make("PandaPush-v3")

model = DDPG(policy="MultiInputPolicy", env=env, replay_buffer_class=HerReplayBuffer, verbose=1)
...
```

## System

  - OS: ubuntu 20.04
  - Python version (`python --version`): 3.8
  - Package version (`pip list | grep panda-gym`):  current master 
  -

qgallouedec, 2022-10-23:
Stable-baselines3 is not yet compatible with gymnasium v0.26 (see the related PR: https://github.com/DLR-RM/stable-baselines3/pull/780). Therefore, you can't use panda-gym v3 with SB3 for the moment. You have to upgrade to panda-gym version 2.0.0.

```shell
pip install panda-gym==2.0.0
```

and use `PandaPush-v2`.

qgallouedec, 2022-10-23:
> The following error occurs on trying to run train_push

I left this file in the repo to anticipate, but we will have to wait a bit for it to be functional.

qgallouedec, 2022-10-23:
Disclaimer added: https://github.com/qgallouedec/panda-gym/commit/09ecfb6d57e0b516ae248ef1d54a4cc7b3db71f1

closed 2022-10-25 by qgallouedec