qgallouedec/panda-gym #39: Problems encountered in manual Stack

learningxiaobai, 2022-10-29:
Hello，when I want to stack in manual:`env = gym.make("PandaStack-v3", render=True)
observation, info = env.reset()

for _ in range(1000):
    current_position = observation["observation"][0:3]
    desired_position = observation["desired_goal"][0:3]
    action = 5.0 * (desired_position - current_position)
    observation, reward, terminated, truncated, info = env.step(action)

    if terminated or truncated:
        observation, info = env.reset()

env.close()`
I met an error:`ValueError: operands could not be broadcast together with shapes (3,) (4,) (4,)`
Thanks!

qgallouedec, 2022-10-30:
Stacking environnement requires actions of shape (4,): displacement (3) and gripper action (1). In your code, action has shape (3,). You need to add 1 element in the action. 
Is the code from the documentation?

learningxiaobai, 2022-10-31:
> Stacking environnement requires actions of shape (4,): displacement (3) and gripper action (1). In your code, action has shape (3,). You need to add 1 element in the action. Is the code from the documentation?

Yes，i will hava a try

learningxiaobai, 2022-10-31:
Still met  ValueError: operands could not be broadcast together with shapes (0,) (4,)：
code：
`
for _ in range(1000):
    current_position = observation["observation"][0:4]
    desired_position = observation["desired_goal"][7:11]
    action = 5.0 * (desired_position - current_position)
    observation, reward, terminated, truncated, info = env.step(action)

    current_position = observation["observation"][7:11]
    desired_position = observation["desired_goal"][23:27]
    action = 5.0 * (desired_position - current_position)
    observation, reward, terminated, truncated, info = env.step(action)

    if terminated or truncated:
        observation, info = env.reset()

env.close()

`

learningxiaobai, 2022-10-31:
Hello，I find it hard to write a example about stack，how to express the gripper action？I will appreciate it if you give me a example about Stack.Thanks.

qgallouedec, 2022-11-01:
It is not easy to write a complete algorithm for the stacking task. The idea is to first go to the first object, grab it, move it without collision, put it down etc.
This is what the first step could look like (quite similar to what you find in the documentation).

```python
import gymnasium as gym
import numpy as np

import panda_gym

env = gym.make("PandaStack-v3", render=True)

observation, info = env.reset()

for _ in range(100):
    gripper_position = observation["observation"][0:3]
    blue_object_position = observation["observation"][7:10]
    position_control = 10 * (blue_object_position - gripper_position)
    gripper_control = np.array([0.1])
    action = np.concatenate((position_control, gripper_control))
    observation, reward, terminated, truncated, info = env.step(action)

    if terminated or truncated:
        observation, info = env.reset()

env.close()
```

learningxiaobai, 2022-11-01:
> It is not easy to write a complete algorithm for the stacking task. The idea is to first go to the first object, grab it, move it without collision, put it down etc. This is what the first step [...]

Thanks，I will try

closed 2022-11-01 by learningxiaobai