qgallouedec/panda-gym #41: Unsupported structured space 'gymnasium.spaces.dict.Dict'

WanqingXia, 2022-11-20:
I am trying to run your code with stable-baselines3 (code under examples->train_push.py), it seems like stable-baselines does not support the observation space structure defined in gymnasium. How am I suppose to solve this problem?

`
Traceback (most recent call last):
  File "/home/wxia612/panda-gym/examples/train_push.py", line 10, in <module>
    model = DDPG(policy="MultiInputPolicy", env=env, replay_buffer_class=HerReplayBuffer, verbose=1)
  File "/home/wxia612/.local/lib/python3.8/site-packages/stable_baselines3/ddpg/ddpg.py", line 85, in __init__
    super().__init__(
  File "/home/wxia612/.local/lib/python3.8/site-packages/stable_baselines3/td3/td3.py", line 103, in __init__
    super().__init__(
  File "/home/wxia612/.local/lib/python3.8/site-packages/stable_baselines3/common/off_policy_algorithm.py", line 111, in __init__
    super().__init__(
  File "/home/wxia612/.local/lib/python3.8/site-packages/stable_baselines3/common/base_class.py", line 179, in __init__
    env = self._wrap_env(env, self.verbose, monitor_wrapper)
  File "/home/wxia612/.local/lib/python3.8/site-packages/stable_baselines3/common/base_class.py", line 228, in _wrap_env
    env = DummyVecEnv([lambda: env])
  File "/home/wxia612/.local/lib/python3.8/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py", line 29, in __init__
    self.keys, shapes, dtypes = obs_space_info(obs_space)
  File "/home/wxia612/.local/lib/python3.8/site-packages/stable_baselines3/common/vec_env/util.py", line 67, in obs_space_info
    assert not hasattr(obs_space, "spaces"), f"Unsupported structured space '{type(obs_space)}'"
AssertionError: Unsupported structured space '<class 'gymnasium.spaces.dict.Dict'>'

Process finished with exit code 1
`

qgallouedec, 2022-11-20:
Duplicate #38 
Please downgrade to `panda-gym==1.1.1`

WanqingXia, 2022-11-21:
Hi! thanks for your reply! I just noticed the declaration on top of the code, by following this issue https://github.com/DLR-RM/stable-baselines3/pull/780. I found a way to not downgrade panda-gym and still make the code works.

first install the fix in stable-baselines3: pip install git+https://github.com/carlosluis/stable-baselines3@fix_tests
then add these three lines at the beginning of python code
```
import sys
import gymnasium
sys.modules["gym"] = gymnasium
```

qgallouedec, 2022-11-21:
Great! Thank you for sharing this.

araffin, 2022-11-22:
@qgallouedec I actually found that hack 6 days ago (https://github.com/DLR-RM/stable-baselines3/pull/780#issuecomment-1315562866) and documented it in the PR: https://github.com/DLR-RM/stable-baselines3/pull/780#issue-1144872152

WanqingXia, 2022-11-23:
> araffin

I must somehow found your solution and didn't realise that. Thanks for sharing!

closed 2022-11-23 by WanqingXia