qgallouedec/panda-gym #42: How to change the object in the task of pick and place

Jay-Vim-Lv, 2022-11-28:
hello, i failed to find whe way when i want to change the object in the task of pickandplace to see the performance.
i noticed that the object of pickandplace task is created by the function create_box (or _create_geometry in fact ) in the pybullet.py
It seemd  i can achived my goal through modefing the self. physics_client.createMultiBody. so i change the code between line 575 and 579,  added the filename where the object (i want to load, obj type) saved. But obviously my mehotd is not correct.
I faild to change the object, so i'd like to ask you how to do that if you have time to reply me.

qgallouedec, 2022-11-28:
I would advise to create you own task, following [the custom env section of the doc](https://panda-gym.readthedocs.io/en/latest/custom/custom_env.html). However, for a minimal effort, you can just modify this lines of PickAndPlace task

https://github.com/qgallouedec/panda-gym/blob/09ecfb6d57e0b516ae248ef1d54a4cc7b3db71f1/panda_gym/envs/tasks/pick_and_place.py#L36-L42

with, for example (if you want a sphere instead)

```python
self.sim.create_sphere(
        body_name="object", 
        radius = self.object_size / 2,
        mass=1.0,
        position=np.array([0.0, 0.0, self.object_size / 2]),
        rgba_color=np.array([0.1, 0.9, 0.1, 1.0]), 
)

Jay-Vim-Lv, 2022-11-28:
OK, thank you very much. l would have a try to create a new task

closed 2022-12-31 by qgallouedec