qgallouedec/panda-gym #43: Too much console output during env creation

jonasreiher, 2022-12-06:
## Describe the bug

Currently, when creating an environment, a whole bunch of stuff is printed to the console:
<details>
<summary>console output</summary>

```
argv[0]=--background_color_red=0.8745098114013672
argv[1]=
argv[2]=
argv[3]=
argv[4]=
argv[5]=
argv[6]=
argv[7]=
argv[8]=
argv[9]=
argv[10]=
argv[11]=
argv[12]=
argv[13]=
argv[14]=
argv[15]=
argv[16]=
argv[17]=
argv[18]=
argv[19]=
argv[20]=
argv[21]=--background_color_green=0.21176470816135406
argv[22]=
argv[23]=
argv[24]=
argv[25]=
argv[26]=
argv[27]=
argv[28]=
argv[29]=
argv[30]=
argv[31]=
argv[32]=
argv[33]=
argv[34]=
argv[35]=
argv[36]=
argv[37]=
argv[38]=
argv[39]=
argv[40]=
argv[41]=
argv[42]=--background_color_blue=0.1764705926179886
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=45
argv[0] = --unused
argv[1] = --background_color_red=0.8745098114013672
argv[2] = 
argv[3] = 
argv[4] = 
argv[5] = 
argv[6] = 
argv[7] = 
argv[8] = 
argv[9] = 
argv[10] = 
argv[11] = 
argv[12] = 
argv[13] = 
argv[14] = 
argv[15] = 
argv[16] = 
argv[17] = 
argv[18] = 
argv[19] = 
argv[20] = 
argv[21] = 
argv[22] = --background_color_green=0.21176470816135406
argv[23] = 
argv[24] = 
argv[25] = 
argv[26] = 
argv[27] = 
argv[28] = 
argv[29] = 
argv[30] = 
argv[31] = 
argv[32] = 
argv[33] = 
argv[34] = 
argv[35] = 
argv[36] = 
argv[37] = 
argv[38] = 
argv[39] = 
argv[40] = 
argv[41] = 
argv[42] = 
argv[43] = --background_color_blue=0.1764705926179886
argv[44] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Mesa/X.org
GL_RENDERER=llvmpipe (LLVM 12.0.0, 256 bits)
GL_VERSION=4.5 (Core Profile) Mesa 21.2.6
GL_SHADING_LANGUAGE_VERSION=4.50
pthread_getconcurrency()=0
Version = 4.5 (Core Profile) Mesa 21.2.6
Vendor = Mesa/X.org
Renderer = llvmpipe (LLVM 12.0.0, 256 bits)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = Mesa/X.org
ven = Mesa/X.org
```
</details>

I couldn't find a way to block this output from outside, since it doesn't seem to use the Python stdout. Otherwise `with contextlib.redirect_stdout(os.devnull):` should work. I also noticed that in IPython only the argvs are output but I have no idea how to achieve this effect via a .py script.

This output seems to partially be PyBullet's fault, the argvs are initialized via panda-gym, though. All empty argvs could be removed here.

## To Reproduce

Provide a **minimal** code :

```python
import gymnasium as gym
import panda_gym

env = gym.make('PandaPush-v3')
```

## System

  - OS: macOS 12.6.1
  - Python version (`python --version`): 3.8.10
  - Package version (`pip list | grep panda-gym`): 3.0.0

qgallouedec, 2022-12-06:
I totally agree with you. Those logs are very anoying. 

> This output seems to partially be PyBullet's faul

They are all from pybullet. However, pybullet devs provide no way to set the verbosity, see https://github.com/bulletphysics/bullet3/issues/3302.

jonasreiher, 2022-12-06:
I see, but I think at least all the empty arguments from the argument list can be removed:

The arguments are created here:
https://github.com/qgallouedec/panda-gym/blob/09ecfb6d57e0b516ae248ef1d54a4cc7b3db71f1/panda_gym/pybullet.py#L28-L32

But due to line breaks and indentation, `options` has a lot of whitespace:
```
'--background_color_red=0.8745098114013672                     --background_color_green=0.21176470816135406                     --background_color_blue=0.1764705926179886'
```

Every space seems to trigger the creation of a new argument. Something like this gets rid of the extra whitespace and therefore significantly shortens the argument list:
```
options = " ".join(options.split())
```
```
'--background_color_red=0.8745098114013672 --background_color_green=0.21176470816135406 --background_color_blue=0.1764705926179886'
```

qgallouedec, 2022-12-06:
Thanks for spotting this! I would prefer  an F-string. I would be happy to receive a PR for this.