qgallouedec/panda-gym #71: pybullet.error: Error loading texture

yoyostudy, 2023-07-27:
Hello,  I try to run the test file test_env using command pytest envs_test.py.

However, it shows some error:

```
FAILED envs_test.py::test_env[PandaFlip-v3] - pybullet.error: Error loading texture
FAILED envs_test.py::test_env[PandaFlipJoints-v3] - pybullet.error: Error loading texture
FAILED envs_test.py::test_env[PandaFlipDense-v3] - pybullet.error: Error loading texture
FAILED envs_test.py::test_env[PandaFlipJointsDense-v3] - pybullet.error: Error loading texture
```

Do you know how to fix that? 

Thank you so much

qgallouedec, 2023-07-27:
What is your OS? What the version you use for panda-gym and pybullet?

yoyostudy, 2023-07-27:
Thanks for your quick response!

I am using Ubuntu, conda environment, python3.8. I guess the panda-gym version I am using is 3.0.7, and the pybullet version is 3.2.5.

```
(rl) yoyo@yoyo:~/Documents/intern/panda-gym/examples$ pip show panda-gym
Name: panda-gym
Version: 3.0.7
Summary: Set of robotic environments based on PyBullet physics engine and gymnasium.
Home-page: https://github.com/qgallouedec/panda-gym
Author: Quentin GALLOUÉDEC
Author-email: gallouedec.quentin@gmail.com
License: 
Location: /home/yoyo/anaconda3/envs/internrl/lib/python3.8/site-packages
Requires: gymnasium, numpy, pybullet, scipy
Required-by: 
```

```
(rl) yoyo@yoyo:~/Documents/intern/panda-gym/examples$ pip show pybullet
Name: pybullet
Version: 3.2.5
Summary: Official Python Interface for the Bullet Physics SDK specialized for Robotics Simulation and Reinforcement Learning
Home-page: https://github.com/bulletphysics/bullet3
Author: Erwin Coumans, Yunfei Bai, Jasmine Hsu
Author-email: erwincoumans@google.com
License: zlib
Location: /home/yoyo/anaconda3/envs/internrl/lib/python3.8/site-packages
Requires: 
Required-by: panda-gym
```

qgallouedec, 2023-07-27:
That's really strange. The CI passes with the exact same configuration: https://github.com/qgallouedec/panda-gym/actions/runs/5676508527/job/15383646275

Have you altered `panda_gym/assets/colored_cubed.png` files by any chance?

Is the following code giving an error? If yes, please provide the full traceback.

```python
import gymnasium as gym
import panda_gym

env = gym.make("PandaFlip-v3")
env.reset()
```

yoyostudy, 2023-07-27:
Hi, I don't think I changed any code or files. And it does return an error:

```
>>> env = gym.make("PandaFlip-v3")
pybullet build time: May 20 2022 19:44:17
argv[0]=--background_color_red=0.8745098114013672
argv[1]=--background_color_green=0.21176470816135406
argv[2]=--background_color_blue=0.1764705926179886
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/yoyo/anaconda3/envs/internrl/lib/python3.8/site-packages/gymnasium/envs/registration.py", line 801, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/yoyo/anaconda3/envs/internrl/lib/python3.8/site-packages/panda_gym/envs/panda_tasks.py", line 53, in __init__
    task = Flip(sim, reward_type=reward_type)
  File "/home/yoyo/anaconda3/envs/internrl/lib/python3.8/site-packages/panda_gym/envs/tasks/flip.py", line 26, in __init__
    self._create_scene()
  File "/home/yoyo/anaconda3/envs/internrl/lib/python3.8/site-packages/panda_gym/envs/tasks/flip.py", line 32, in _create_scene
    self.sim.create_box(
  File "/home/yoyo/anaconda3/envs/internrl/lib/python3.8/site-packages/panda_gym/pybullet.py", line 451, in create_box
    texture_uid = self.physics_client.loadTexture(texture_path)
pybullet.error: Error loading texture
```

Thank you

qgallouedec, 2023-07-27:
It seems like the texture for the coloured cube can't be loaded in your config. Can you download the texture:

https://github.com/qgallouedec/panda-gym/blob/master/panda_gym/assets/colored_cube.png

and then execute the following:

```python
import pybullet_utils.bullet_client as bc

client  = bc.BulletClient()
client.loadTexture("/path/to/colored_cube.png")
```

closed 2023-10-24 by qgallouedec

ChenWEIx2, 2024-02-26:
check the assets folder in your anaconda path (/anaconda3/envs/env_name/lib/python3.8/site-packages/panda_gym/assets), you might find that missing the "colored_cube.png" file,   add it under the assets folder.