qgallouedec/panda-gym #74: how to realize the sb3 train demo code with panda-gym?

NUSLCC, 2023-08-19:
Tried and failed with panda-gym==2.0.0 as required, gym==0.26.2, stable-baseline3==2.1.0
Can anyone provide a clue of how to run the demo [train](https://panda-gym.readthedocs.io/en/latest/usage/train_with_sb3.html#train) code, thanks!!
Also tried to downgrade the gym==0.21.0 by #19, still failed.
[image]

qgallouedec, 2023-08-19:
The documentation was not up to date. It should be fixed now. Use the following code:

```python
import gymnasium as gym
import panda_gym
from stable_baselines3 import DDPG

env = gym.make("PandaReach-v3")
model = DDPG(policy="MultiInputPolicy", env=env)
model.learn(30_000)
```

NUSLCC, 2023-08-19:
Thanks! It works in my environment now.

closed 2023-08-19 by NUSLCC