qgallouedec/panda-gym #77: Render screen is out of sync

Qingliang-Li, 2023-10-09:
## Describe the bug

After i=14, the mechanical arm does not continue to move in the render screen, and the screen stops. But render continues to render, and Render's picture is not consistent with the actual robotic arm motion.

## To Reproduce

Provide a **minimal** code :

```python
import gymnasium as gym
import panda_gym
import matplotlib.pyplot as plt
from matplotlib import animation
import numpy as np

def display_frames_as_gif(frames):
    patch = plt.imshow(frames[0])
    plt.axis('off')

    def animate(i):
        patch.set_data(frames[i])

    anim = animation.FuncAnimation(plt.gcf(), animate, frames=len(frames), interval=5)
    anim.save('panda.gif', writer='pillow', fps=30)

frames = []
# env = gym.make("PandaReach-v3", render_mode="human")
env = gym.make("PandaReach-v3", render_mode="rgb_array", renderer="OpenGL")
observation, info = env.reset()

# 生成圆的参数
theta = np.linspace(0, 2 * np.pi, 1000)
r = 0.1  # 圆的半径
# 计算圆的坐标
x = r * np.cos(theta)
y = r * np.sin(theta)
z = np.zeros_like(theta) + observation["observation"][2]  # 所有点都在z=0的平面上
pre_tarj = np.zeros((1000, 3))
pre_tarj[:, 0] = x
pre_tarj[:, 1] = y
pre_tarj[:, 2] += observation["observation"][2]  # 所有点都在z=0的平面上
tarj = np.zeros_like(pre_tarj)
# 初始化位置
desired_position = pre_tarj[0, :]

for i in range(100):
    # Render into buffer.
    image = env.render()
    frames.append(image)
    print(i)
    current_position = observation["observation"][0:3]
    desired_position = pre_tarj[i, :]
    tarj[i, :] = current_position

    action = 5.0 * (desired_position - current_position)
    observation, reward, terminated, truncated, info = env.step(action)

    # if terminated or truncated:
    #     observation, info = env.reset()

env.close()
display_frames_as_gif(frames)
```

## System

  - OS: Win 10
  - Python version (`python --version`): python 3.11
  - Package version (`pip list | grep panda-gym`):

qgallouedec, 2023-10-12:
I can't reproduce the issue. Here is the screen recording of OpenGL GUI (with the correct fps):

https://github.com/qgallouedec/panda-gym/assets/45557362/0f1d1ac9-9fbb-4342-a25c-eefd5536bb06

And the gif generated (also with the correct fps):

[image]

Qingliang-Li, 2023-10-12:
> I can't reproduce the issue. Here is the screen recording of OpenGL GUI (with the correct fps):
> 
>  Screen.Recording.2023-10-12.at.12.10.43.mov 
> And the gif generated (also with the correct fps):
> 
> [image] [ [image] ](https://user-images.githubusercontent.com/45557362/274564983-ee7957b8-13e2-4f33-a359-ae9b648637ef.gif) [ ](https://user-images.githubusercontent.com/45557362/274564983-ee7957b8-13e2-4f33-a359-ae9b648637ef.gif)

If you add a few more frames to the GIF, you will notice that the robot arm stops moving in the later frames. In this video, you can observe that the robotic arm remains stationary during the second, third, and fourth seconds, although it should be in motion.

Qingliang-Li, 2023-10-12:
If you add a few more frames to the GIF, you will notice that the robot arm stops moving in the later frames. In this video, you can observe that the robotic arm remains stationary during the second, third, and fourth seconds, although it should be in motion.

qgallouedec, 2023-10-12:
It moves slowly (because the desired position you've defined evolves slowly), but it's not stationary.

qgallouedec, 2023-10-12:
If you change a bit your desired positions, here is what you get:

```python
import gymnasium as gym
import panda_gym
import matplotlib.pyplot as plt
from matplotlib import animation
import numpy as np

def display_frames_as_gif(frames):
    patch = plt.imshow(frames[0])
    plt.axis("off")

    def animate(i):
        patch.set_data(frames[i])

    anim = animation.FuncAnimation(plt.gcf(), animate, frames=len(frames), interval=5)
    anim.save("panda.gif", writer="pillow", fps=25)

env = gym.make("PandaReach-v3", render_mode="rgb_array", renderer="OpenGL")
observation, info = env.reset()

z = observation["observation"][2]

theta = np.linspace(0, 20 * np.pi, 1000)
r = 0.1
desired_positions = np.stack([r * np.cos(theta), r * np.sin(theta), z * np.ones(1000)], axis=1)

frames = []
for i in range(200):
    frames.append(env.render())
    current_position = observation["observation"][0:3]
    desired_position = desired_positions[i]
    action = 5.0 * (desired_position - current_position)
    observation, reward, terminated, truncated, info = env.step(action)

env.close()
display_frames_as_gif(frames)
```
[image]

Qingliang-Li, 2023-10-12:
I get it. Thank you very much，Gallouedec.

closed 2023-10-13 by qgallouedec