qgallouedec/panda-gym #81: Shape of `achieved_goal` and `desired_goal`

behradkhadem, 2023-11-20:
Hi. I'm trying to make my own environment using this package for my own robot with its own task. Things to consider:

- I'm using TQC algorithm.
- I'm defining my own task, which uses the position of end-effector inside its reward (`compute_reward` method).

My reward function looks like this:
```
    def compute_reward(self, achieved_goal, desired_goal, info: Dict[str, Any]) -> np.ndarray:
        d = distance(achieved_goal, desired_goal)
        ee_position = self.sim.get_link_position(self.body, 11)
        # if achieved_goal.shape != ee_position.shape:
        #    print(f'Shapes are not same! {achieved_goal.shape}, {ee_position.shape}, {desired_goal.shape}')
        ee_d = distance(achieved_goal, ee_position)

        distance_penalty = 10 * d + ee_d 
        # print(f'reward: {-1 * distance_penalty}')
        return -(distance_penalty).astype(np.float32)

```
The problem is that when I'm trying to calculate the distance between `achieved_goal` and `ee_position` I get this error that these 2 don't have the same shape; the first is (1638, 3) and the latter is (3,). Why the shape of  `achieved_goal` and `desired_goal` is not (3,)? And how can I solve this issue?

behradkhadem, 2023-11-21:
> However, the `compute_reward` function is called during the replay phase

After disabling `HerReplayBuffer` I saw that it's training correctly and you're right on that. Right now I have 2 files: One for the robot (implementing `PyBulletRobot`) and one for the task (implementing `Task`). Which one should I edit? From what I understood, everything would be fine if I edit `get_achieved_goal` method inside the task file. Am I right?

Thanks in advance.

qgallouedec, 2023-11-21:
If I understand correctly, you want to create a reward that depends on 
1. the object's position relative to the end-effector
2. the object's position relative to the desired position

A kind of dense pick and place.

I've implemented it for panda:

```python
import numpy as np

from panda_gym.envs.core import RobotTaskEnv, Task
from panda_gym.envs.robots.panda import Panda
from panda_gym.pybullet import PyBullet
from panda_gym.utils import distance

class PickAndPlaceDense(Task):
    def __init__(self, sim, get_ee_position):
        super().__init__(sim)
        self.get_ee_position = get_ee_position
        self.object_size = 0.04
        with self.sim.no_rendering():
            self._create_scene()

    def _create_scene(self):
        self.sim.create_plane(z_offset=-0.4)
        self.sim.create_table(length=1.1, width=0.7, height=0.4, x_offset=-0.3)
        self.sim.create_box(
            body_name="object",
            half_extents=np.ones(3) * self.object_size / 2,
            mass=1.0,
            position=np.array([0.0, 0.0, self.object_size / 2]),
            rgba_color=np.array([0.1, 0.9, 0.1, 1.0]),
        )
        self.sim.create_box(
            body_name="target",
            half_extents=np.ones(3) * self.object_size / 2,
            mass=0.0,
            ghost=True,
            position=np.array([0.0, 0.0, 0.05]),
            rgba_color=np.array([0.1, 0.9, 0.1, 0.3]),
        )

    def get_obs(self):
        object_position = self.sim.get_base_position("object")
        object_rotation = self.sim.get_base_rotation("object")
        object_velocity = self.sim.get_base_velocity("object")
        object_angular_velocity = self.sim.get_base_angular_velocity("object")
        return np.concatenate([object_position, object_rotation, object_velocity, object_angular_velocity])

    def get_achieved_goal(self):
        ee_position = np.array(self.get_ee_position())
        object_position = np.array(self.sim.get_base_position("object"))
        achieved_goal = np.concatenate([ee_position, object_position])
        return achieved_goal

    def reset(self):
        obj_goal = self._sample_goal()
        self.goal = np.concatenate([obj_goal, obj_goal])  # goal is [object_goal, ee_goal], and we want them to be the same
        object_position = self._sample_object()
        self.sim.set_base_pose("target", obj_goal, np.array([0.0, 0.0, 0.0, 1.0]))
        self.sim.set_base_pose("object", object_position, np.array([0.0, 0.0, 0.0, 1.0]))

    def _sample_goal(self):
        low = np.array([-0.15, -0.15, self.object_size / 2])
        high = np.array([0.15, 0.15, 0.2 + self.object_size / 2])
        return self.np_random.uniform(low, high)

    def _sample_object(self):
        low = np.array([-0.15, -0.15, self.object_size / 2])
        high = np.array([0.15, 0.15, self.object_size / 2])
        return self.np_random.uniform(low, high)

    def get_distance_penalty(self, achieved_goal, desired_goal):
        ee_position = achieved_goal[:3]
        object_position = achieved_goal[3:]
        object_goal = desired_goal[3:]
        dist_object_to_goal = distance(object_position, object_goal)
        dist_ee_to_object = distance(ee_position, object_position)
        distance_penalty = 10.0 * dist_object_to_goal + dist_ee_to_object
        return distance_penalty

    def is_success(self, achieved_goal, desired_goal):
        d = self.get_distance_penalty(achieved_goal, desired_goal)
        return np.array(d < 0.05, dtype=bool)

    def compute_reward(self, achieved_goal, desired_goal, info):
        distance_penalty = self.get_distance_penalty(achieved_goal, desired_goal)
        return -(distance_penalty).astype(np.float32)

class PandaPickAndPlaceDenseEnv(RobotTaskEnv):
    def __init__(self, render_mode: str = "rgb_array") -> None:
        sim = PyBullet(render_mode=render_mode)
        robot = Panda(sim, base_position=np.array([-0.6, 0.0, 0.0]))
        task = PickAndPlaceDense(sim, get_ee_position=robot.get_ee_position)
        super().__init__(robot, task)

env = PandaPickAndPlaceDenseEnv(render_mode="human")

observation, info = env.reset()

for _ in range(1000):
    action = env.action_space.sample()
    observation, reward, terminated, truncated, info = env.step(action)
    if terminated or truncated:
        observation, info = env.reset()
```

behradkhadem, 2023-11-21:
Thanks, but I still get the same error while using replay buffer. 
```
[~/anaconda3/envs/SB3/lib/python3.9/site-packages/sb3_contrib/tqc/tqc.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/anaconda3/envs/SB3/lib/python3.9/site-packages/sb3_contrib/tqc/tqc.py) in learn(self, total_timesteps, callback, log_interval, tb_log_name, reset_num_timesteps, progress_bar)
    300         progress_bar: bool = False,
    301     ) -> SelfTQC:
--> 302         return super().learn(
    303             total_timesteps=total_timesteps,
    304             callback=callback,

[~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/common/off_policy_algorithm.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/common/off_policy_algorithm.py) in learn(self, total_timesteps, callback, log_interval, tb_log_name, reset_num_timesteps, progress_bar)
    329                 # Special case when the user passes `gradient_steps=0`
    330                 if gradient_steps > 0:
--> 331                     self.train(batch_size=self.batch_size, gradient_steps=gradient_steps)
    332 
    333         callback.on_training_end()

[~/anaconda3/envs/SB3/lib/python3.9/site-packages/sb3_contrib/tqc/tqc.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/anaconda3/envs/SB3/lib/python3.9/site-packages/sb3_contrib/tqc/tqc.py) in train(self, gradient_steps, batch_size)
    207         for gradient_step in range(gradient_steps):
    208             # Sample replay buffer
--> 209             replay_data = self.replay_buffer.sample(batch_size, env=self._vec_normalize_env)  # type: ignore[union-attr]
    210 
    211             # We need to sample because `log_std` may have changed between two gradient steps

[~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/her/her_replay_buffer.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/her/her_replay_buffer.py) in sample(self, batch_size, env)
    221         real_data = self._get_real_samples(real_batch_indices, real_env_indices, env)
    222         # Create virtual transitions by sampling new desired goals and computing new rewards
--> 223         virtual_data = self._get_virtual_samples(virtual_batch_indices, virtual_env_indices, env)
    224 
    225         # Concatenate real and virtual data

[~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/her/her_replay_buffer.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/her/her_replay_buffer.py) in _get_virtual_samples(self, batch_indices, env_indices, env)
    311 
    312         # Compute new reward
--> 313         rewards = self.env.env_method(
    314             "compute_reward",
    315             # the new state depends on the previous state and action

[~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py) in env_method(self, method_name, indices, *method_args, **method_kwargs)
    125         """Call instance methods of vectorized environments."""
    126         target_envs = self._get_target_envs(indices)
--> 127         return [getattr(env_i, method_name)(*method_args, **method_kwargs) for env_i in target_envs]
    128 
    129     def env_is_wrapped(self, wrapper_class: Type[gym.Wrapper], indices: VecEnvIndices = None) -> List[bool]:

[~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/anaconda3/envs/SB3/lib/python3.9/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py) in <listcomp>(.0)
    125         """Call instance methods of vectorized environments."""
    126         target_envs = self._get_target_envs(indices)
--> 127         return [getattr(env_i, method_name)(*method_args, **method_kwargs) for env_i in target_envs]
    128 
    129     def env_is_wrapped(self, wrapper_class: Type[gym.Wrapper], indices: VecEnvIndices = None) -> List[bool]:

[~/projects/mobile-panda/envs/mobile_panda_v1/new_mobile_panda.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/projects/mobile-panda/envs/mobile_panda_v1/new_mobile_panda.py) in compute_reward(self, achieved_goal, desired_goal, info)
     85 
     86     def compute_reward(self, achieved_goal, desired_goal, info):
---> 87         distance_penalty = self.get_distance_penalty(achieved_goal, desired_goal)
     88         return -(distance_penalty).astype(np.float32)
     89 

[... cut ...]
     79         distance_penalty = 10.0 * dist_object_to_goal + dist_ee_to_object
     80         return distance_penalty

[~/projects/panda-gym/panda_gym/utils.py](https://vscode-remote+wsl-002bubuntu.vscode-resource.vscode-cdn.net/home/behradx/projects/mobile-panda/envs/mobile_panda_v1/~/projects/panda-gym/panda_gym/utils.py) in distance(a, b)
     12         np.ndarray: The distance between the arrays.
     13     """
---> 14     assert a.shape == b.shape
     15     return np.linalg.norm(a - b, axis=-1)
```

qgallouedec, 2023-11-21:
My bad, you actually need:

```diff
-         ee_position = achieved_goal[:3]
+         ee_position = achieved_goal[..., :3]
-         object_position = achieved_goal[3:]
+         object_position = achieved_goal[..., 3:]
-         object_goal = desired_goal[3:]
+         object_goal = desired_goal[..., 3:]
```

I've modified my initial code.

behradkhadem, 2023-12-02:
Sorry for delay in answering. I finally got it working and your solution worked. 
I haven't tested fully (for large number of training episodes) but it seems that there won't be any issues there. If you're ok with it, I'll keep this issue open until final test. 

Thanks for helping.

closed 2023-12-02 by qgallouedec