qgallouedec/panda-gym #84: Decreasing Size of Action Space

behradkhadem, 2023-12-24:
I'm creating a custom environment for my own robot (that has total 12 degrees of freedom). I want to disable some of my joints to see its effect in training duration and quality; but no matter how much I've tried I've failed to do that and I have to put all of my joints in `joint_indices`, otherwise I'll get errors (from simulation layer). 

Is it possible to disable some of my joints without editing the URDF file?

qgallouedec, 2023-12-24:
I'd probably do something like that:

```python
import numpy as np
from gymnasium import spaces

from panda_gym.envs.core import PyBulletRobot

class MyRobot(PyBulletRobot):
    def __init__(self, sim):
        action_dim = 2 # = number of joints to control (2/4)
        action_space = spaces.Box(-1.0, 1.0, shape=(action_dim,), dtype=np.float32)
        super().__init__(
            sim,
            body_name="my_robot",  # choose the name you want
            file_name="my_robot.urdf",  # the path of the URDF file
            base_position=np.zeros(3),  # the position of the base
            action_space=action_space,
            joint_indices=np.array([0, 2, 3, 5]),  # list of the indices, as defined in the URDF
            joint_forces=np.array([1.0, 1.0, 1.0, 1.0]),  # force applied when robot is controled (Nm)
        )

    def set_action(self, action):
        action = np.concatenate([action, np.zeros(2)]) # control only 2 joints out of 4
        self.control_joints(target_angles=action)

    ...
```

Tell me if it works.

behradkhadem, 2023-12-25:
I finally did it! The issue was solved by changing the joint types from `revolute` and `prismatic` to `fixed`. I think it was a PyBullet issue.

closed 2023-12-25 by behradkhadem