qgallouedec/panda-gym #85: Compatibility of gym-0.24 branch?

selamie, 2024-01-31:
Hi there, 

Completely understand if this branch is deprecated and not guaranteed to be supported, but I wanted to ask if the gym-0.24 branch is in a working state and compatible with gym-0.24 or earlier versions. I cannot get `env = gym.make("PandaReach-v2", render=True)` to work for example, and it returns the error `AttributeError: 'PandaReachEnv' object has no attribute 'observation_space'`

I wondered if the authors might comment on the state of this branch and whether it was intended to be usable with any of the pre-gymnasium versions of gym, or any known issues I could potentially fix myself to use with older gym environments. 

Thank you! 
--Selam

qgallouedec, 2024-02-02:
Hi,
No this branch is not supposed to work. I think it was a transition branch when we went from gym 0.21 to gym 0.26. Anyway, if possible I suggest you use either gym 0.21 or gymnasium.

selamie, 2024-02-02:
Thank you, I was able to download an older version from the "releases" section and got that working instead!

closed 2024-02-02 by selamie