qgallouedec/panda-gym #86: panda-gym v2 error; "TypeError: catching classes that do not inherit from BaseException is not allowed".

ppangTae, 2024-02-28:
## Describe the bug

The docs said that panda-gym v3 is not compatible with SB3, so I tried to use v2.
So I installed it with "pip install panda-gym==2.0.4" and ran the code as shown in the docs and got the following error "TypeError: catching classes that do not inherit from BaseException is not allowed".

## To Reproduce

Provide a **minimal** code :

The code below is what I ran.

import gym
import panda_gym
from stable_baselines3 import DDPG

env = gym.make("PandaReach-v2")
model = DDPG(policy="MultiInputPolicy", env=env)
model.learn(30_000)

## System

  - OS: Window 11
  - Python version (`python --version`): 3.9.10
  - Package version (`pip list | grep panda-gym`): 2.0.4(panda-gym) (I had previously installed v3, and then reinstalled v2.)

ChiaChing-Yen, 2024-06-05:
Try adding `env.close()` at the end of the script?

see #80

qgallouedec, 2024-06-10:
I'd recommend using v3 and gymnasium
Yes the error originates from pybullet when you don't close the env (as far as I remember which shouldn't be an issue if you close the close)
If the error persists with v3, please open a new issue

closed 2024-06-10 by qgallouedec