qgallouedec/panda-gym #87: Clarification about 'observation', 'achieved_goal' and 'desired_goal'

wilhem, 2024-03-18:
I'm trying to implement a simulation environment using Panda-Gym, but I couldn't find enough information.
I'm still confused about the difference among the following: _observation["observation"]_, _observation["achieved_goal"]_ and _observation["desired_goal"]_ .
Initially I thought, that _observation["observation"]_ contains the joint values of the pandas.
But  then I saw this example [here](https://panda-gym.readthedocs.io/en/latest/usage/manual_control.html), where the current_position takes the first 3 elements of the _observation["observation"]_. That's means, that the _observation["observation"][0:3]_ contains the x, y, z position of the grip. Is it right?
Then I have the question about the _observation["achieved_goal"]_. What is this exactly _achieved_goal_ ? Does it contain the last achieved goal and gets updated once a new goal has been reached? Or it is the x, y, z position of the grip?

Many thanks

wilhem, 2024-03-24:
Hi,
thank you very much for your answer. I found the following [paper](https://arxiv.org/pdf/2106.13687.pdf) (linked by you) very interesting.
But there is a point, which makes me unsure: you wrote the following:

> For Reach, there is no component linked to the task because there is no object (in this case, the terminology is a bit misleading because there is still a task to perform, but this is a special case so we left it like that)

How should I understand that statement? Is the `desired_goal` array NOT containing the position of the object to be reached?

qgallouedec, 2024-03-27:
> How should I understand that statement? Is the `desired_goal` array NOT containing the position of the object to be reached when using the `Reach` environment?

This is a design-related remark.
I wanted to dissociate the task from the robot. Thus, the criterion for achieving the task should not depend on the robot's state. For example, if the task is to push an object to a target position, the only thing that matters is whether the object's position matches the desired position. For Reach, this is special because the task involves the robot's state, but it's meant to be a special case. 
For observation, `desired_goal` is the gripper's target position (to be precise, it's a position, not an object to be reached), `achieved_goal`, the gripper's position.

> By the way: is the list of [observable features](https://github.com/qgallouedec/panda-gym/issues/8#issuecomment-911512499) still valid? Or it has changed? The list is vor `-v1` and now we have `-v3`

Yes

wilhem, 2024-03-27:
> > How should I understand that statement? Is the `desired_goal` array NOT containing the position of the object to be reached when using the `Reach` environment?
> For Reach, this is special because the task involves the robot's state, but it's meant to be a special case. For observation, `desired_goal` is the gripper's target position (to be precise, it's a position, not an object to be reached), `achieved_goal`, the gripper's position.

Sorry again. But this last statement did confuse me completely.
I'm using the _Reach_ task. In that case the gripper should reach a green object on the working table. A reward of 0.0 is granted when the difference between gripper and object less than 5 cm is.
Among 'observation', 'achieved_goal', 'desired_goal', which one:

1. is the position of the green object?
2. is the position of the gripper?
3. are the first 3 values of 'observation' the actual position of the gripper?

qgallouedec, 2024-03-27:
> Sorry again. But this last statement did confuse me completely.

Don't worry, if you're asking the question, it's not clear enough.

> I'm using the Reach task. In that case the gripper should reach a green object on the working table

Yes, two details though,
- the green thing is just for visualisation, the robot cannot interact with it. 
- it's not necessary on the table, it can be anywhere above.

> A reward of 0.0 is granted when the difference between gripper and object less than 5 cm is.

True

> which one:
> 1. is the position of the green object?

`desired_goal`

> is the position of the gripper?

`achieved_goal`

> are the first 3 values of 'observation' the actual position of the gripper?

Yes

closed 2024-06-10 by qgallouedec