qgallouedec/panda-gym #88: Flip and Stack question

Swk6666, 2024-04-01:
The environments PandaStack-v3 and PandaFlip-v3, whether there are feasible results for them, I saw in the paper that the success rate of the stack is almost 0.

qgallouedec, 2024-04-01:
Both are feasible, in the sense that technically, you can write your hard-coded policy that would solve the task.
However, stack has an highly spare reward, and flip required a very precise control. Both are very challenging task, but possible.

Swk6666, 2024-04-01:
Thank you for your reply. Are there any algorithms/hyperparameters that have already been successful, or do I need to implement them myself?

NicholasCorrado, 2024-04-15:
@Swk6666 I haven't tried PandaStack-v3, but I was able to solve PandaFlip-v3 with reasonable accuracy using the DDPG implementation in stable-baselines3.
Appendix G in [this paper](https://arxiv.org/pdf/2310.17786) lists the hyperparameters I used.

closed 2024-06-10 by qgallouedec