qgallouedec/panda-gym #95: Set to move in a two-dimensional plane

ChenyangRan, 2024-06-14:
Hi, I'm using the panda-gym to create my UR3 Reach. But when I set the initial pos, and set_action(np.array(0,0,0)), the ur3 switch its pos automatically. It seems to automatically align tcp with xyz in the environment. Could you please tell me how to fixed the initial pos and just move in 2D?

Before:
[image]

 set action is [0,0,0]
[image]

qgallouedec, 2024-06-14:
No 100% to understand. 0,0,0 is the position reached by the ee in the second image?

ChenyangRan, 2024-06-14:
> No 100% to understand. 0,0,0 is the position reached by the ee in the second image?

No, I want to do nothing. But the ee change its position.

qgallouedec, 2024-06-14:
How have you set the neutral position? With `self.neutral_joint_values`? https://github.com/qgallouedec/panda-gym/blob/746ee4d89ecf1d17b14d22512d4cc2497d10ba49/panda_gym/envs/robots/panda.py#L45

ChenyangRan, 2024-06-14:
> self.neutral_joint_values = np.array([0.00, 0.41, 0.00, -1.85, 0.00, 2.26, 0.79, 0.00, 0.00])

I just have tested the neutral_joint_values, but it still hapeended.
Start
[image]

After
[image]

qgallouedec, 2024-06-14:
can you share the code of your robot?

ChenyangRan, 2024-06-14:
It's ok. Thank you very much.
[upload.zip](https://github.com/user-attachments/files/15838484/upload.zip)